Welthulk/Sparlectra.jl #19: @JuliaRegistrator register

Welthulk, 2024-03-29:
@JuliaRegistrator register

JuliaRegistrator, 2024-03-29:
Registration pull request updated: [JuliaRegistries/General/103677](https://github.com/JuliaRegistries/General/pull/103677)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.4.8 -m "<description of version>" e53f73af0c0dd6ee26f470d7b8b5f07118adab0e
git push origin v0.4.8
```
Also, note the warning: This looks like a new registration that registers version 0.4.8.
Ideally, you should register an initial release with 0.0.1, 0.1.0 or 1.0.0 version numbers
This can be safely ignored. However, if you want to fix this you can do so. Call register() again after making the fix. This will update the Pull request.

Welthulk, 2024-03-30:
done

closed 2024-03-30 by Welthulk