Welthulk/Sparlectra.jl #25: @JuliaRegistrator register

Welthulk, 2024-03-30:
@JuliaRegistrator register

## Version 0.4.10 (2024-05-01)

### New Features
 - removed numerous redundant functions, partially restructured classes
 - removed support for CGMES due to the availability of numerous alternative libraries
 - removed support for the legacy custom JSON data format (potentially subject to reintroduction at a later stage)
 
### Enhancements
- added functions to facilitate the creation of networks
- better performance

### Bug Fixes
- calculation of branch flow and losses
- branches out of service

JuliaRegistrator, 2024-03-30:
Registration pull request updated: [JuliaRegistries/General/103923](https://github.com/JuliaRegistries/General/pull/103923)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.4.10 -m "<description of version>" 033fd040a3ea0aaf193cd4fc5dd97151bc03675e
git push origin v0.4.10
```
Also, note the warning: Version 0.4.10 skips over 0.4.9
This can be safely ignored. However, if you want to fix this you can do so. Call register() again after making the fix. This will update the Pull request.

Welthulk, 2024-03-30:
[merge approved]

Welthulk, 2024-03-30:
nothing happend after registration of new features....

Welthulk, 2024-04-02:
@JuliaRegistrator register

JuliaRegistrator, 2024-04-02:
Registration pull request updated: [JuliaRegistries/General/103923](https://github.com/JuliaRegistries/General/pull/103923)

### Tip: Release Notes

Did you know you can add release notes too? Just add markdown formatted text underneath the comment after the text
"Release notes:" and it will be added to the registry PR, and if TagBot is installed it will also be added to the
release that TagBot creates. i.e.

```
@JuliaRegistrator register

Release notes:

## Breaking changes

- blah
```

To add them here just re-invoke and the PR will be updated.

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.4.10 -m "<description of version>" 715db7c0f56e1e748b7dbc86485360cdffd7e533
git push origin v0.4.10
```
Also, note the warning: Version 0.4.10 skips over 0.4.9
This can be safely ignored. However, if you want to fix this you can do so. Call register() again after making the fix. This will update the Pull request.

Welthulk, 2024-04-02:
done, thx

closed 2024-04-02 by Welthulk