Welthulk/Sparlectra.jl #46: Hotfix wrong function call

sudo-ac, 2024-04-14:
@JuliaRegistrator register()

Release notes:

## Version 0.4.17 (2024-04-14)
### New Features
 - added testcase for importing Matpower files

### Bug Fixes
 - bugfix wrong function call in `createnet_powermat`

JuliaRegistrator, 2024-04-14:
Registration pull request created: [JuliaRegistries/General/104882](https://github.com/JuliaRegistries/General/pull/104882)

### Tagging

After the above pull request is merged, it is recommended that a tag is created on this repository for the registered package version.

This will be done automatically if the [Julia TagBot GitHub Action](https://github.com/marketplace/actions/julia-tagbot) is installed, or can be done manually through the github interface, or via:
```
git tag -a v0.4.17 -m "<description of version>" 4ddfe51f4de6b8068ab77ed92948b6506f208ef5
git push origin v0.4.17
```

sudo-ac, 2024-04-14:
closed with release 0.4.17

closed 2024-04-14 by sudo-ac